DDeGonge/OS-ARM #2: bill of materials for the non-3D printed parts

3mrrrx, 2021-05-25:
bill of materials for the non-3D printed parts would be very very very very helpful recreate and try the project.

thank you!

DDeGonge, 2021-06-05:
I'll be filling in the BOM as I work on the project, apologies for the delay. The harmonic drive uses 608 and 688 bearings, plus some M3 square nuts and M3 bolts (I believe 20mm length for the bearing posts). But I will be uploading a full BOM in the next few weeks!

jlogan223, 2023-02-26:
Is there any update on this?

DDeGonge, 2023-03-10:
> Is there any update on this?

Hey sorry about that. I'm currently reworking the design from scratch so this design of the full arm is frozen. I plan to have a full BOM with the redesign which will be in Onshape so it is more accessible. But I don't have any idea of the full BOM for this iteration so if you want to build any part of it, you will need to open the cad and work out fasteners from that. Bearings are already included. And some heat stake inserts are used in the larger shallow holes where there's a screw but no obvious nut.