DDeGonge/OS-ARM #7: Motor coupling missing?

Eufrades, 2022-03-25:
Hello DDeGonge,
I like the design of your 75mm harmonic drive, and I want to use it to drive a sand feeder for my sand spreader on the back of my garden tractor. 
I printed all of the .stl parts from the V2 folder, but the wave generator is not the same as the one in your video, its obviously a newer version. There appears to be a coupling of some sort that goes between the wave generator and the motor. I've looked to see if I have missed a file, but I can't find it. Is that hiding somewhere?

Thanks for all of your work,
Eufrades

gandrewstone, 2022-03-28:
I think its meant to take a 20T GT2 timing belt pulley with the top flange popped off (based on a comment to the video here: https://www.youtube.com/watch?v=Emvo3bLT-Z4&t=16s).  I've got a stepper out of an old epson printer with a pulley that seems to match the wave generator... except I'm not sure because not sure how to pop the top flange off (blowtorch maybe?)

Since you are active, I have a question for you... have you figured out how the top housing is secured?  Seems to me like it would just fall off

gandrewstone, 2022-03-29:
Thats a good thought about the large bearing.  I bought one so in a few days I'll be able to tell.  That probably won't work for me either.  I was thinking of using it for a SCARA robot, so there will be force parallel to the motor shaft.  Maybe a stationary lid with a thrust bearing between will work for me, or something through the axle... What is your application?

Anyway the gear i had didnt actually fit, off by 2mm, so i made a bigger hole.  I also had elephant feet on my flex spline, so i fixed that by raising the teeth a tiny bit off the bed and printing with no supports.  I also put the skirt on the inside only.  I can give you the bkender models if you want.

gandrewstone, 2022-03-29:
What about the 688 bearings?  I think that they were originally on the motor shaft, but are not needed in the V2 design.

Eufrades, 2022-03-29:
I think you're right on the bearings, it doesn't look like they're needed anymore. As long ans the motor is mounted solidly to the bottom housing.
My application isn't typical for this type of gearbox, I just needed something with a really big speed reduction in a small package. I'm planning to use it on a sand spreader on the back of my atv. There is a motor there that already spins the spreader, but I wanted to use the same motor to really slowly turn an agitator in the sand hopper because the sand keeps getting hung up in there.
I appreciate the offer of your files, but I think I'm going to modify this quite a bit more to slow it down even more for me.