DDeGonge/OS-ARM #8: Reduction ratio

Okosai, 2022-11-17:
So, i am modifying v2 harmonic drive (for my motorised telescope mount) and i am trying to figure out what is actual rediction ratio for it. As far as i understood one of the rings has 70 teeth and 68 another, meanwhile spline has 66 teeth... could not find any info on that.
Thanks

KoehIl180151, 2023-01-27:
I would also need this information.
I tried to calculate it myself with the steps per rev and i got 1:29 but in testing it didn't quite work out.

bfra2373, 2023-01-27:
Looking at [wiki](https://en.wikipedia.org/wiki/Strain_wave_gearing) You should see that the  ratio is 
[image]
or For example, if there are 202 teeth on the circular spline and 200 on the flex spline, the reduction ratio is (200 − 202)/200 = −0.01

DDeGonge, 2023-01-27:
Because my harmonic drive is designed different than the standard, the formula above needs slight modification.

The number of teeth in my "flex spline" is irrelevant, instead what matters is the number of teeth on the input (circle spline equivalent) vs output (flex spline equivalent) sides of the housing. So the formula would be (68 - 70) / 68, which gives a reduction of **34:1**. Note this equation results in a negative number, which means the output spins the opposite direction of the input.

closed 2023-01-27 by DDeGonge